Severson-Group/AMDC-Hardware #35: Consider power sequencing to fix inrush current issues

npetersen2, 2018-09-27:
Reduced bulk capacitance on VIN to 1000uF. This solved PicoZed "blinky" issue (reduced inrush current).

"If I have the powerstack connected when I power the AMDC up, the inrush from the powerstack seems to still be causing blinky. May need to implement some sort of power sequencing on rev 3. I'll think about it" -Jason.

This issue serves as a reminder for this problem.

jjsylvestre, 2018-10-01:
Powered up board, then connected 5V5 jumper per Eric's recommendation. Same results - blinky

jjsylvestre, 2018-10-01:
Inrush current on 5V5 node
[image]

jjsylvestre, 2018-10-01:
Solved problem by adding 2ohm resistor inline with 5V5 node to limit inrush current

[image]

npetersen2, 2018-10-01:
@jjsylvestre Can you decide how we should fix this on REV C and file a bug issue describing what should be done (and close this issue)? Thanks.

jjsylvestre, 2018-10-02:
See Issue #43 for continued conversation on this

closed 2018-10-02 by jjsylvestre