Severson-Group/AMDC-Hardware #42: Too much inrush current on 5V5 from Picozed

jjsylvestre, 2018-10-02:
Proposed solution is current limiting resistor with SPST relay in parallel that close from an enable signal from the PicoZed

jjsylvestre, 2018-10-02:
[image]

npetersen2, 2018-11-13:
Talked with @jjsylvestre:

Would like the PZ_ENABLE signal in above picture to not be controlled from software, but instead in h/w.

Our idea: use a RC circuit with time constant of ~1 sec, feeding into a comparator, which then drives a n-channel MOSFET to enable relay, thus shorting out inrush limiting resistor.

I'll implement this idea and post again for discussion.

npetersen2, 2018-11-14:
**Proposed implementation:**
[image]

Need to pick relay to use, add flyback diode, and add LED to indicate when resistor is inline.

@jjsylvestre thoughts?

npetersen2, 2018-11-15:
Thinking that main resistor should be 2 ohms (not 1), due to power supply limit of 3A at 5.5V.

This would limit current to 2.75A, and require a (2.75)^2 * 2 = ~15W resistor... Pretty beefy.

jjsylvestre, 2018-11-15:
Looks good.  Looks like comparator is TTL output. Are you 100% certain that when IN+ > IN-, that corresponds to a high output? Couldn't find that info in the ds. 

As for the power value of the resistor, we can probably some get away with something lower since the current is exponentially decaying with time. 

Also, we might want to take a look at the 5V rail during the startup event to see how stable it is.

npetersen2, 2018-11-16:
- Changed comparator to one that specified its output level vs. input voltages (LMV761MAX/NOPB)
- Added flyback diode
- Using four 2512 SMD 6.8R resistors in parallel to achieve high max power, while being cheap and relatively small footprint.

[image]

npetersen2, 2018-11-16:
Last thing is to check 5V rail during startup to see that it is stable enough to use for cap charging / timing. Will update issue with results.

npetersen2, 2018-11-16:
Plot of 24V board input (C2) vs. 5V rail (C1) during startup. Looks good to me.

[image]

npetersen2, 2018-11-17:
Reopening for investigation on not using mechanical relay. Maybe use BJT? Currently looking up ANs for ideas. 

See this article about why FPGA startup is typically an issue for power supplies:
http://www.ti.com/lit/an/slyt079/slyt079.pdf

npetersen2, 2018-11-17:
This paper describes how to design MOSFET-based inrush current limiter:
http://www.mosaic-industries.com/embedded-systems/_media/pdfs/application-notes/motorola-an1542-active-inrush-current-limiter.pdf

npetersen2, 2018-11-17:
Final implementation: using TPS22965 (http://www.ti.com/lit/ds/symlink/tps22965.pdf), costs $1.

This is a 5.7-V, 6-A, 16-mΩ On-Resistance Load Switch. Basically just an n-channel MOSFET which has an adjustable slew rate for turn on.

This slew rate determines inrush current. I think the only capacitance on the 5.5V node is the bypass caps for the LDOs. Therefore, this is 50uF. I calculated values resulting in inrush of ~700mA.

[image]

closed 2018-11-17 by npetersen2