Severson-Group/AMDC-Hardware #52: Update Inrush Current Limiter circuitry

npetersen2, 2018-12-11:
The calculated value for C48 was too small and still resulted in the hiccuping issue.

I stacked a 100nF cap on top of C48 on the REV C board and it fixed the issue.

We probably don't need to increase C48 all the way up to 100nF from 100pF, but it does work...

Let's update the schematics to reflect this needed change.

[image]

elsevers, 2018-12-11:
If the datasheet of this part requests a 100pF capacitor for reasons of ESR, etc., we may consider placing a 100 nF capacitor in parallel with the 100pF capacitor.

npetersen2, 2019-09-01:
Done.

closed 2019-09-01 by npetersen2