Severson-Group/AMDC-Hardware #68: Analog input op amp failed

npetersen2, 2019-05-16:


elsevers, 2019-11-29:
@Nick-Hemenway Believes that this has now happened on three AMDCs

elsevers, 2019-11-29:
On the AMDC that @Anvar-Khamitov is debugging: the -15V rail is collapsed down to 0V (-15V_A is collapsed, -15V_B appears to be okay, as are all of the positive rails).

elsevers, 2019-11-29:
@ngadiyar93, @Anvar-Khamitov, and @mimayberry report that in a separate AMDC, a couple of the op amps get hot and the voltage rails occasionally collapse. Can one of you three can provide more information on this. Specifically, which power rails are collapsing (`-15V_A`, `-15V_B`, `+15V_A`, `+15V_B`)? Which op amps are getting hot?

elsevers, 2019-11-29:
This seems like something we need to debug. Let's come up with a plan. 

One strategy: change the ADC sample rate down to a low value (like <10 kHz) or completely disable the ADC sampling and see if the issue goes away (for the AMDC where the power rails intermittently collapse). How hard is it to make the change? Does this require an update to verilog?

Anvar-Khamitov, 2019-12-02:
> On the AMDC that @Anvar-Khamitov is debugging: the -15V rail is collapsed down to 0V (`-15V_A` is collapsed, `-15V_B` appears to be okay, as are all of the positive rails).
> 
> @Anvar-Khamitov, can you please reply to this with the serial number of this AMDC, so that we can keep the boards straight. Also, please add this issue to the board log for this AMDC.

Serial number is 05. The issue is added to the board log.

elsevers, 2019-12-02:
Please confirm that in all of these boards it is the `ADA4522` parts that are overheating/failing (not the INA2143` parts).

elsevers, 2019-12-02:
On the boards where the power rails intermittently collapse: 
- what does the analog signal look like at the input and output of the `ADA4522` parts (when the rail is collapsed/is not collapsed)
- Erase the FPGA/DSP and confirm that the ADCs are not being sampled (verify that `ADC_CNV`, `ADC_SCK`, and `ADC_CLKOUT` are not active). Do the parts still get hot and do the voltage rails collapse?

Anvar-Khamitov, 2019-12-02:
> Please confirm that in all of these boards it is the `ADA4522` parts that are overheating/failing (not the `INA2143` parts).

@elsevers Based on the results using thermal camera, `ADA4522` parts are overheating, not `INA2143`. This can be seen in the image below for the board with the serial number 03: `ADA4522` are the ones with light yellow color, there are 4 of them. There are 2 `INA2143`-s under each `ADA4522`. 
But we are not sure whether overheating of `ADA4522` means that it is failing, not `INA2143`.

[image]

Anvar-Khamitov, 2019-12-02:
> On the boards where the power rails intermittently collapse:
> 
> * what does the analog signal look like at the input and output of the `ADA4522` parts (when the rail is collapsed/is not collapsed)? Do we have some kind of high frequency system level noise that is making these buffer op amps work too hard?
> * Erase the FPGA/DSP and confirm that the ADCs are not being sampled (verify that `ADC_CNV`, `ADC_SCK`, and `ADC_CLKOUT` are not active). Do the parts still get hot and do the voltage rails collapse?

@elsevers This board (serial number 03) has another issue. When connected to power supply, it starts beeping. According to @npetersen2 : "one of the DC/DC converters on the board is "hiccupping" ie it turns on, current limits, and turns itself off. This keeps repeating and is causing the beeping. This could mean either the power converters on AMDC are broken, or a component is broken, thus drawing way too much current from the DC/DC."

Due to this issue, I am not sure whether we can do tests with this board.

elsevers, 2019-12-03:
> @elsevers This board (serial number 03) has another issue. When connected to power supply, it starts beeping. According to @npetersen2 : "one of the DC/DC converters on the board is "hiccupping" ie [...]

I have created an issue #82 for you to track your debugging progress with the 5V supply / 2.048V reference problem with SN 03.

mimayberry, 2019-12-06:
The `-15V_A` rail on board 05 is now operational. The problem was an incorrectly placed voltage regulator. A +15V regulator was installed in place of the correct `-15V_A` regulator. The part in question is shown below:  
[image]

elsevers, 2019-12-09:
Any update on who built this board? Was it Screaming Circuits?

mimayberry, 2019-12-09:
@elsevers The board log only says it was made by Beta.
[image]

elsevers, 2019-12-09:
Beta paid for Screaming Circuits to build those boards.

Anvar-Khamitov, 2019-12-23:
All 15 V voltage rails of AMDC with serial number 5 are collapsed: `-15V_A`, `-15V_B`, `+15V_A`, `+15V_B` are collapsed down to -1.84 V, -1.9 V, -0.93 V, -0.6 V, respectively. Started to debug this issue with @ASHAD1 . Possible cause might be ESD or AMDC overloading the op-amps.

ASHAD1, 2019-12-26:
The Analog conditioning circuitry of AMDC serial 5 failed again, we had replaced the ICs and its working now. These might be damaged because of ADC overloading (current spikes) the Op-amps as mentioned before in the previous comment, or this might happen because of ESD as mentioned by @Anvar-Khamitov

elsevers, 2019-12-27:
@ASHAD1 and @Anvar-Khamitov what parts did you have to replace? ADA4522 or the INA2143? And did you have to replace all of them or only certain ones?

My understanding is that these parts have not had to be replaced on this board previously (SN05). Is that correct?

ASHAD1, 2019-12-27:
@elsevers  we had replaced four INA2143 (U47C, U48C, U47D, U48D), all the ADA4522 opAmps and +15V regulators and -15V regulators.

These parts were not replaced previously on the board SN05.

elsevers, 2019-12-27:
@ASHAD1 Are you certain that all of these parts had failed? This is an important detail for debugging purposes: the INA2143 parts are not loaded by the ADCs (only the ADA4522 parts are) -- so if both have failed, this would seem to suggest either an ESD issue or some kind of systemic issue.

elsevers, 2019-12-28:
@ASHAD1 Has promised a write up with more debug information. It sounds like the ADA4522 parts failed initially and then the INA2143 parts failed during debugging.

@ASHAD1 and @Anvar-Khamitov: Do we have any evidence that the ADC’s sampling is causing the ADA4522's to warm up? Has anyone tried turning the ADC’s off and seeing if the ADA4522's cool off, then turning the ADC’s on and seeing if they warm back up? (This could be done by erasing the flash on the picozed and then using a probe to verify that the ADC’s are not active). *Can someone run this test on board SN05*... right away? 

Further thoughts:

- The datasheet for the ADC shows an implementation where a unity gain OP Amp is used with a RC low pass filter between the op amp and the ADC: https://www.analog.com/media/en/technical-documentation/data-sheets/ltc2320-12.pdf (see Fig. 4 and 6). Note that the LT1819 part is suggested as a unity gain ADC driver: https://www.analog.com/media/en/technical-documentation/data-sheets/18189fb.pdf

- This application note talks about RC filters between the OP Amp and the ADC being a good idea to attenuate the "kick" from the ADC sampling: https://www.analog.com/en/analog-dialogue/articles/front-end-amp-and-rc-filter-design.html# 

- When we designed the analog signal chain, we intentionally did not place an RC filter between the op amp and the ADC because we felt that the unity gain op amp would give us a stiff supply

- The ADA4522 is a fancy op amp and it is not marketed as an ADC driver: https://www.analog.com/media/en/technical-documentation/data-sheets/ADA4522-1_4522-2_4522-4.pdf . It has thermal and ESD protection circuitry. These sound like good things, but I wonder if they may add some unexpected behavior when interacting with the "kick" of the ADC load capacitance changing. The data sheet also talks about potential instability when driving capacitive loads and the suggested remedy is adding ~20 ohm of output resistance (though, note that the part claims to be able to drive ~250 pF of capacitive load without instability, and the ADC is only ~15 pF....)

ASHAD1, 2019-12-28:
Yes, ADA4522 was damaged initially and during debugging INA2143 were damaged. The debugging done by me and @Anvar-Khamitov  are as follows.
1. Initially, one +15V regulator and one ADA4522 opamp was very hot (>90c), so we removed them and tested the voltage.
2. After that all other +-15 V were back on. One ADA opAmp had -15V rail voltage but no +15V (As we had removed one regulator). This Op-Amp got very hot and was damaged, I think because the input was more than the positive supply.
3. Then we removed the recently damaged OpAmp and also added +15V regulator, now all the +-15V were back on. But while measuring I accidentally touched one +-15V rail and I think at that instant there was negative voltage at positive rail damaging 4 INA, 2 opAmps and +15V regulator.
4. After replacing the damaged IC, the board was working.

I checked the board with ADC and also by removing the supply to the ADC, the result was interesting, there was no significant difference in the temperature in both the cases, it was around 37C after 5 min.

@elsevers had also mentioned that there is a possibility of ADA4522 output railing to positive supply (+15V in our design) for using high input resistance, as per figure 82 of datasheet. We have used relatively small input resistance which might avoid this condition, but I think we have to improve this in future. Further suggestions required.

I was also checking the application notes for op-amp interfacing with ADC. In all these notes they have used RC filter between opAmp and ADC.
https://www.digikey.com/en/articles/techzone/2017/sep/find-the-right-low-noise-amplifier-for-precision-adcs
https://www.analog.com/media/en/training-seminars/design-handbooks/Op-Amp-Applications/Section3.pdf
https://www.ti.com/lit/an/sloa098/sloa098.pdf

elsevers, 2019-12-28:
@ASHAD1 I am confused by the results of the thermal test you describe here. You cut the power to the ADC but the ADA4522 is still heating up...

- Where is the current path? when the ADC does not have voltage supplies what do its analog input rails appear as (you can check the data sheet)? They may function like a forward bias diode to a power rail, effectively creating a short circuit on the ADA4522's output. 

- Did you use the oscilloscope to check the part to see if it is being excited by some high frequency noise or is trying to drive the ADC to an unreasonable voltage? 

Please try your test again, but this time have the ADC powered up, but inactive (by either erasing the flash on the picozed or using Nathan's firmware to make it inactive)

If you have a proposal for changes to the board, please make it asap. **Be specific:** if you want to stop using the ADA4522's, please pick out a replacement part from Digikey.  *For example, we can consider replacing the ADA4522 with a 0-5V part like you have mentioned and/or we can consider putting a 1st order RC filter between the unity gain amplifier and the ADC -- if we decide it doesn't help us, we can put a 0 ohm resistor and not populate the capacitor.*

elsevers, 2019-12-29:
@Anvar-Khamitov @ASHAD1  I understand that you have done an additional test where the picozed was removed, but the ADA4522 was still hot. I have two questions that I would like you to respond to this morning as I am having lunch with an ADC expert:

1. Approximately what temperature was the ADA4522?

2. What was the ADA4522 trying to output? A DC voltage? AC voltage? What voltage level?

*Please answer within the hour if possible*

Anvar-Khamitov, 2019-12-29:
@elsevers When picozed was removed (verified that there is no ADC clock signal), we got these results:
1. The ADA4522 temperature was approximately 38-39 C (the same as when picozed was not removed)
2. ADA4522 output was a DC voltage with around 2.05 V (almost equal to V_REF). Compared to the case when picozed was not removed, this DC voltage Is much smoother and has much less noise.

Anvar-Khamitov, 2019-12-29:
Might not be quite clear from these pictures, but here are the plots of ADA4522 output voltage with and without picozed:

[image]
[image]

elsevers, 2019-12-29:
Thanks Anvar. What are the volts/div?

Any ideas what that small amount of noise is on the amplifier output when the picozed is unplugged?

[with_picozed]<https://user-images.githubusercontent.com/42312168/71560422-d26abd80-2a37-11ea-8cf5-d4fadb98a1e6.JPG>
[without_picozed]<https://user-images.githubusercontent.com/42312168/71560423-d696db00-2a37-11ea-8123-93ed23c8597a.JPG>

Anvar-Khamitov, 2019-12-29:
@elsevers  Below are some clearer plots. All these plots have 100 mV/div and 2 us/div (in the previous plots, it was 500 ns/div). 

First photo shows ADA4522 voltage when picozed is plugged in. Second is when picozed is unplugged - it turns out that this noise is periodic. It seems to be ambient noise because this noise in the output of ADA4522 is still present even when there is no supply to the AMDC (third photo).

[image]
[image]
[image]

elsevers, 2019-12-30:
Thanks @Anvar-Khamitov. This was very helpful.

The ADA4522 is intended to operate as a unity gain amplifier. Its purpose was to give us a solid voltage source for the ADC to read from. However, it does not seem to be doing a great job of this, as evidenced in the plots above, and furthermore is periodically failing. We do not yet know why it is failing or why it is heating up (note that the part is in fact rated for infinite duration short circuit conditions.)

**I think we should consider removing the ADA4522/unity gain amplifier stage from Rev D.** This means that the output of the RC filter will connect directly through to the ADC pin. The ADC switches a 15 pF load in and out for each approximation (which seems to be causing the voltage spikes on the ADA4522 waveform). The RC filter is currently using a 33nF capacitor, which should certainly be stiff enough against the 15 pF load switching in and out.

It would be a good idea to validate that removing the ADA4522 results in working hardware before we make the change to Rev D. We can do this by removing the ADA4522 parts and soldering a jumper/short between the in+/in- pin pair of each amplifier (i.e., short pin 2 to pin 3; short pin 5 to pin 6; short pin 9 to pin 10; short pin 12 to pin 13). 

- @Anvar-Khamitov and @ASHAD1 is this something you can test (_removing ADA4522s and adding shorts_) by the January 1st deadline that @npetersen2 has given us for board changes?

- I would like to see some LTSpice simulations validating the analog signal chain. I'll create this as a separate issue. 

- Anyone have comments / feedback on my proposed change here?

Anvar-Khamitov, 2019-12-30:
I can try to test it today (removing ADA4522 parts).

npetersen2, 2019-12-30:
@Anvar-Khamitov @elsevers Were these tests performed with voltage applied to the input of the analog signal chain, or was it floating? I'm not sure how the system behaves when no input is connected.

@Anvar-Khamitov when you remove the ADA4522 part, can you try hooking up a reference input to the signal chain (maybe just constant 1V etc) and then probe that input plus any other signals of interest?

Anvar-Khamitov, 2019-12-30:
@npetersen2 There were no analog inputs, the voltage was floating when we performed these tests. Ideally, when there is 0 V input, ADA4522 output should be V_REF, but it was at around 2.059 V (mean of the voltage in the above pictures).

I can test with 1 V input after removing ADA4522.

Anvar-Khamitov, 2019-12-31:
@elsevers @npetersen2 After removing ADA4522s and adding shorts, it looks like there are no voltage spikes in the output signal due to ADC, as shown in the first photo (there is no input to the analog chain). Looks like it now has only ambient noise.

When 1 V is applied to the input of the analog chain, the output signal is shown in the second photo. Third photo shows the waveform of the input analog signal (1 V).

[image]
[image]
[image]

elsevers, 2019-12-31:
Thanks @Anvar-Khamitov. Very helpful to see these images. I am curious about that ~200 kHz noise that occurs when the AMDC is powered off...

I have proposed issue #88 to be implemented in REVD to address our problems here.

@Anvar-Khamitov, can you test this further by regulating closed loop currents when your analog signal chain is set up without the ADA4522 part?

[65099D90-24FE-4C4E-8F5E-046A1BF962B9]<https://user-images.githubusercontent.com/42312168/71607930-01ba2100-2b43-11ea-987d-bc4eebf1f4d8.jpeg>
[034E7B98-3FF4-4E13-8DDE-8805D46FC219]<https://user-images.githubusercontent.com/42312168/71608074-f74c5700-2b43-11ea-9b13-b04b19154b97.jpeg>
[7D0F0765-C688-4021-8CA8-2B6DA1E990F5]<https://user-images.githubusercontent.com/42312168/71608078-fe736500-2b43-11ea-8c76-48458f9973ea.jpeg>

Anvar-Khamitov, 2019-12-31:
@elsevers I will be able to test it with closed loop current control on Thursday.

Anvar-Khamitov, 2020-01-02:
Here are some plots from the closed loop current regulation (no ADA4522 parts). I tested it for two cases. 

Photo 1 shows reference and actual currents with reference current amplitude 1 A and frequency 50 Hz. Photo 2 is the output from INA2143 (input to ADC).

Photo 3 shows reference and actual currents with reference current amplitude 1.5 A and frequency 100 Hz. Photo 2 is the output from INA2143 (input to ADC).

[image]
[image]
[image]
[image]

closed 2020-01-29 by npetersen2