Severson-Group/AMDC-Hardware #82: 2.048V reference rails failed on AMDC Rev C SN03

elsevers, 2019-12-03:
I am creating an issue to track this.

AMDC Serial Number 03 has a problem where both of the 2.048V reference voltages are collapsed to nearly 0 V and one of the ADCs is burning hot. 

Prior to this, @Anvar-Khamitov and @mimayberry observed that the 15V rails would randomly exhibit issue #68, but the op amps would recover. Is it possible that this is all some related systemic problem?

@Anvar-Khamitov and @mimayberry -- please use this issue to track your debugging status for this board.

mimayberry, 2019-12-04:
After removing the two 2.048V regulators, the 5V and 2.048V rails remain shorted to ground. The removed regulators appear to undamaged.
Without the regulators, the 5.5V supply continues to beep and the 5V rail drops to 4.7V.

mimayberry, 2019-12-06:
Removing both ADCs solved the beeping problem. Both components had internal shorts from `Vref` (2.048V) to ground. 

For the +15V rail problem:
On analog channel A, removing both `ADA4522` parts had no effect on the `+15V_A` rails. Then, removing one `INA2143` IC brought `+15V_A` back to the correct voltage. This likely means the `INA2143` part went bad, although the cause is still unknown. The same procedure was repeated for channel B, now all of the +/-15V rails are operational on board 03.

closed 2020-01-29 by npetersen2