Severson-Group/AMDC-Hardware #86: ADG3308 (encoder signal chain) has improper power rails on REV C

elsevers, 2019-12-20:
ADG3308 has improper power rails. The datasheet states that `VCCA` should always be less than `VCCY`, but on AMDC REVC `VCCA` is connected to 5V and `VCCY` is connected to 1.8V. This will cause problems.

Has this already been corrected for REVD?

elsevers, 2019-12-31:
@npetersen2 reports that this part will not be used on REVD. Can this issue be closed?

Anvar-Khamitov, 2020-01-02:
I checked whether I can directly connect MAX3097E (U5) to the picozed inputs by removing ADG3308 (U4) (REV C AMDC). Seems that I cannot do this because maximum input to picozed from the encoder chain is 2.35 V (VCCO + 0.55 V where VCCO = 1.8 V), while the output of MAX3097E is at around 3.5 V.

npetersen2, 2020-01-14:
This should be solved on REV D design. Will review at today's meeting, then close this issue.

closed 2020-01-14 by npetersen2