Severson-Group/AMDC-Hardware #102: Add jumper for selecting external vs. default E-STOP and VDRIVE

npetersen2, 2020-03-04:
In the current REV D design, the PWM inverter interface will not function at all without wires plugged into the external screw connectors for `VDRIVE` and `ESTOP`.

Please add three-pin selectors for the following:

1. E-STOP: One position is external (functions like now), other position is enabled.
2. VDRIVE: One position is external (functions like now), other position is 5V.

npetersen2, 2021-02-25:
Added jumper for VDRIVE input:

[image]

Previously, VDRIVE connected straight to CON2 pin 1. Now, it goes through the header with the option of connecting board 5V directly, thus not needing to use the external pins.

npetersen2, 2021-02-25:
## E-STOP

### Before:

[image]

### After:

[image]

closed 2021-02-25 by npetersen2

GnomeTek, 2021-03-21:
Approved