Severson-Group/AMDC-Hardware #103: isoSPI 24V power doesn't work

npetersen2, 2020-03-24:
The isoSPI connectors are supposed to have 24V power to supply the external circuitry if needed. Indeed, the `VIN2` signal is routed to pins at the connector, but there is an issue.

The `VIN2` is 24V, but in reference to the `VIN-` of the AMDC... NOT the common `GND`. This is due to the isolated DC/DC converters! Therefore, the 24V supply at the isoSPI connectors is NOT usable.

This may be okay since the 5V rail is still available for powering external boards.

[image]

npetersen2, 2021-02-08:
Per planning meeting, this issue should be solved. We want 24V available for the expansion port.

To solve, I propose adding another small isolated DC/DC, probably 6W rated is fine; 24V at 250mA:
https://www.digikey.com/en/products/detail/cui-inc/PQQ6W-Q24-S24-S/13402043

This would add ~$10 to BOM and take up board space, but I think it is worth it and an easy solution.

npetersen2, 2021-02-27:
Implemented:
1. Added 24V DC/DC to form isolated 24V rail
2. Used this new 24V rail for the GPIO port

[image]

closed 2021-02-27 by npetersen2

GnomeTek, 2021-03-21:
Approved