Severson-Group/AMDC-Hardware #137: Fix board start-up PWM issues

npetersen2, 2020-08-10:
The Xilinx Zynq-7000 SoC is used to drive the PWM outputs on the AMDC. During start-up (and any time the FPGA has no bitstream loaded), I believe the pin outputs are random -- they float. This causes strange issues for PWM outputs...

I propose we change the hardware to default the PWM outputs to a known state when the FPGA is not actively driving them. This would usually require something like a pull down resistor on each output... The AMDC has 48 PWM outputs... that would be a lot of resistors!

However, the AMDC has hardware E-STOP support! When the E-STOP is activated, all PWM outputs go to the logic LOW state. This is exactly what we want during start-up. So, to implement this, we just need to AND the E-STOP signal with a digital output from the FPGA. There would just be one pull down on the output from the FPGA. This would cause all PWMs to start in a logic LOW state.

See this crude drawing, where the orange is the new hardware we need (AND gate and one pull-down resistor):

[image]

npetersen2, 2021-02-04:
Should be an OR in the orange logic gate

npetersen2, 2021-02-25:
I have implemented this:

1. Added pin-out from PicoZed called `LV_PS_DRIVE_EN`. When this pin is logic LOW, the drive should be disabled. Used PS_MIO44, which is a PS pin (i.e. connected to DSP, not FPGA). This saves us one pin from the FPGA since this pin won't toggle very fast, so having only DSP control of it is fine. I confirmed in the datasheets for the PicoZed that MIO Bank 1/501 is 1.8V logic, and PS_MIO[40-51] are on Bank 501, therefore, this pin is 1.8V logic.
2. Renamed old `DRIVE_EN` signal to be `ESTOP_DRIVE_EN` as it is 5V logic enable signal derived from ESTOP input.
3. Passed both `ESTOP_DRIVE_EN` and `LV_PS_DRIVE_EN` into `DriveInterface.SchDoc`
4. Added 1.8V to 5V level shift in `DriveInterface.SchDoc` for the `LV_PS_DRIVE_EN`
5. Added AND gate for `PS_DRIVE_EN` and `ESTOP_DRIVE_EN`
6. The final 5V logic `DRIVE_EN` signal then goes to all PWM drivers.

[image]

closed 2021-02-25 by npetersen2

GnomeTek, 2021-03-22:
Circuit implemented as described. Approved!