Severson-Group/AMDC-Hardware #141: Remove isoSPI functionality from the GPIO ports

npetersen2, 2020-08-12:
This is a "large" decision, but I propose we completely remove isoSPI from the AMDC platform. isoSPI is great for low bandwidth communication (~1MHz SPI SCLK freq max), but for high-speed, real-time motor drives, this is not helpful. In fact, isoSPI is generally useless for us.

If we remove all isoSPI, we can then add two extra I/O to each GPIO expansion port; one diff input and one diff output. This would mean we have a 3-line I/O for each expansion port.

This would create compatibility issues with existing hardware, but I think this is the right approach. We can then stop changes the GPIO expansion port pin out.... I believe the only boards that would have to change are the DAC board and the SensorCard platform motherboard.

We should address this before REV E hardware design release.

@elsevers @ASHAD1 

NOTE: If each GPIO port had 3 outputs and 3 inputs, then we could actually add two extra GPIO ports to the AMDC. This would mean we have four HD15 connectors (2 stacks of 2) for GPIO. Each HD15 port would have 3 inputs and 3 outputs. I think this is better than keeping isoSPI in the design.

npetersen2, 2021-02-26:
Implemented this issue:

1. Remove all isoSPI-related circuitry
2. Doubled the GPIO ports due to extra FPGA I/O available since SPI is no longer used

The isoSPI is now called GPIO. There are 2 stacks of 2 HD15 connectors, totaling 4 GPIO ports. Each port has 3 inputs and 3 outputs. All signals are differential. The signal naming is: `GPIO[1..4]_IN[1..3]` for inputs, `GPIO[1..4]_OUT[1..3]` for outputs.

[image]

closed 2021-02-26 by npetersen2

psinha25, 2021-03-22:
@npetersen2 This looks good to me. Signals names (in terms of renaming) are correct. I also compared the provided schematic image above to the `uAMDC` schematics and things correct!