Severson-Group/AMDC-Hardware #146: Fix ESD protection on DAC board

mimayberry, 2020-10-29:
The TVS diodes used for ESD protection keep the DAC from outputting correct voltages  when ±7.5 V outputs are commanded on each set of four channels. 

The current TVS diode contains a center node, that allows for differential voltage to be applied the diodes. This leads to the diodes conducting and dropping the voltage at the DAC output. The current TVS diode looks like this:
[image]

Solution is to use a TVS diode without the center node. Like this: 
[image]

npetersen2, 2022-02-24:
New part: https://www.digikey.com/en/products/detail/smc-diode-solutions/SMDA12CTR/6236252

codecubepi, 2022-04-12:
@npetersen2 I saw that you merged DAC REV C into develop. Can we close this issue now as well, since it was addressed in REV C? The linked PR didn't seem to auto-close it.

PR #204 (2022-03-16): DAC Board: Fix ESD protection by updating TVS diodes
Update U5 and U6 TVS diodes on AMDC DAC board.

Previous diodes were [SMDA12C-4TR](https://www.digikey.com/en/products/detail/smc-diode-solutions/SMDA12C-4TR/6022740?s=N4IgTCBcDaIM4FsAmBDAjGAxgWgCwBcAnEAXQF8g) TVS diodes with a common node. New diodes are [SMDA12CTR](https://www.digikey.com/en/products/detail/smc-diode-solutions/SMDA12CTR/6236252) diodes without the common node.

The reason for this change is that the old common node TVS diodes began to conduct and drop voltage at the DAC output when differential voltage was applied the diod [... cut]
closed 2022-04-19 by npetersen2