Severson-Group/AMDC-Hardware #156: Change shield termination 0R resistors to connect to VIN-, not the board GND layer

npetersen2, 2021-02-08:
Per discussion with @GnomeTek, we do not want the shield noise to be injected to the AMDC board common GND.

Instead, it should be shunted to the VIN ground. This can be accomplished by changing the connector termination resistors to connect to `VIN_GND` via a 1206 0R resistor and a thick trace width. Run the trace on the bottom layer and keep the resistors on the bottom layer as well.

Keep the default for these resistors as DNP.

npetersen2, 2021-02-27:
Implemented:
- Switch to 1206 footprints for resistors
- Keep resistors as DNP
- Change resistor termination from `GND` to `GND_IN`

[image]

GnomeTek, 2021-03-21:
With this being routed to the GND_In like you have now, I suggest placing the DNP as 0-Ohm so they are connected.

npetersen2, 2021-03-25:
Updated all shield termination to be 0R by default, not DNP.

See 95d763e78095244112f9fd5328cd76a998e2168f

closed 2021-03-25 by npetersen2