Severson-Group/AMDC-Hardware #157: Add PTC resettable fuse inline with PWM outputs for short-circuit protection

npetersen2, 2021-02-08:
Per discussion with @GnomeTek, we should protect the PWM outputs from short-circuit / shoot-through currents. When the outputs of the PWM drivers short, it tends to cause the driver IC to fail.

Suggestion is to add a 0603 inline footprint to each PWM signal output line which should fit in the available board space.

The 0603 footprint will be populated by default with the $0.14 ea PTC resettable fuse: https://www.digikey.com/en/products/detail/bel-fuse-inc/0ZCM0004FF2G/10270530

Users can swap this out for a 0R resistor if they need more drive current, which is highly unlikely. The PTC resettable fuse supports up to ~100mA current and has a nominal resistance of 4 ohms. In nominal operating conditions, the output current is usually < 1mA...

npetersen2, 2021-02-25:
1. Created model of part and footprint.
2. Added inline to all PWM outputs:

[image]

closed 2021-02-25 by npetersen2

GnomeTek, 2021-03-21:
Approved!