Severson-Group/AMDC-Hardware #158: Merge UART and JTAG into one interface using Digilent JTAG-SMT3-NC device

npetersen2, 2021-02-08:
The current hardware has two USB ports, one for UART and one for JTAG.

The JTAG module we are using has a new version which integrates the UART functionality, therefore only requiring one USB port. It exposes the UART interface to the host via a virtual COM port.

Please update the schematics and layout for this new interface, and delete all prior UART circuitry.

https://www.digikey.com/en/products/detail/digilent-inc/410-357-B/8605091

npetersen2, 2021-02-26:
Fairly major changes required to implement this issue:

1. Modeled two new P/Ns: `JTAG-SMT3-NC` and USB TYPE B receptacle
2. Created new schematic sheet which implements the USB connection to the JTAG-SMT3-NC device. It exports the JTAG and UART signals, plus a reset line.
3. Deleted the old JTAG sheet and UART sheet
4. Used the new sheet outputs to replace the old sheets (i.e. hooked the new sheet up to the PicoZed).

Overall, this simplifies the schematics, reduces part count by a lot, and provides a single host interface via USB for both JTAG and UART. Great!

[image]

npetersen2, 2021-02-26:
Definitely need reviews on this before ordering boards, but I'm confident in the design.

BrandtBuchda, 2021-03-19:
The `PS_SRST_B` seems to be missing a pull up resistor. Aside from that, it all looks correct according to the datasheets.

psinha25, 2021-03-22:
@npetersen2 I agree with @BrandtBuchda's comment above, everything looks correct based on the datasheet of the module. I believe on page 4 of the datasheet, there is a `10K` pull up resistor on the `PS_SRST_B` pin.

GnomeTek, 2021-03-22:
I agree w/ the others, looks good besides the pullup

npetersen2, 2021-03-25:
Reviewed datasheet per comments above and did the following:
1. Changed the series resistor by the reset button to be 100 ohms, down from 10k.
2. Reviewed PicoZed hardware design and found that no pull-up is included on the PicoZed for the `PS_SRST_B` signal
3. Determined that the pull-up voltage should be 1.8V
4. Add 10k pull-up to 1.8V on `PS_SRST_B` signal in the `JTAG_UART.SchDoc` sheet

closed 2021-03-25 by npetersen2