Severson-Group/AMDC-Hardware #160: Add E-STOP logic signal as input to FPGA

npetersen2, 2021-02-09:
On the current hardware, when the user activates the E-STOP, all PWM outputs turn off (i.e. logic LOW). However, there is no indication of this event to the firmware. This creates a potentially scary situation if the e-stop is then unasserted and the control code was still running.

Please add a single pin input to the FPGA for the e-stop logic signal. This way, an interrupt can occur in the firmware so the user can take action!

npetersen2, 2021-02-25:
1. Added 5V to 1.8V level shifter from the `ESTOP_DRIVE_EN` to `LV_ESTOP_DRIVE_EN` signal
2. Connected `LV_ESTOP_DRIVE_EN` to PS_MIO49
3. Confirmed PS_MIO49 is 1.8V logic
4. Confirmed we can enable interrupts on PS_MIO49 to detect ESTOP button press

closed 2021-02-25 by npetersen2

GnomeTek, 2021-03-21:
Approved!

*did not track down #4 confirming ISR capable input.

psinha25, 2021-03-22:
Looks good to me @npetersen2