Severson-Group/AMDC-Hardware #163: Power Protection Circuit Modifications

GnomeTek, 2021-02-24:
These updates will allow proper operation of the input protection circuit. 

Update resistance values
- R83 - 0 Ohm/ Remove
- R55 - 390K ohm
- R58 - 4.12K ohm
- R59 - 5.9K ohm

Add filter capacitors (surface mount ceramic works fine)
- 56nF from OV pin to GND
- 56nF from UV pin to GND

Add pullup resistor from !FAULT to VIN for debugging. Fault is a pull down only. 
- 100K Ohm

npetersen2, 2021-02-25:
Implemented the required changes. I used 100nF filter caps, hopefully that is ok, @GnomeTek?

[image]

@GnomeTek can you provide the equations you used to compute the resistor values? Or the script, etc. I would like to update the note in the schematics:

[image]

If at all possible, I would love construct the resistance values from just a few 10k and 100k P/Ns that are already in the BOM. Could do 2 10k in parallel to get ~5k. Then, 4 100k in series to get ~400k. Could replace the SHDN pullup with 2 100k. This would remove ~4 lines from the BOM! @GnomeTek thoughts?

npetersen2, 2021-02-25:
Updated filter caps to 33nF per @GnomeTek comments.

npetersen2, 2021-02-26:
Script from @GnomeTek for computing resistor values:

```MATLAB
clc;
close all;
clear all;
​
%% AMDC Curr Vals
R3 = 11.8e6;        % R55 & R83 = 5.9Meg each 11.8Meg 
R2 = 121e3;         % R58 = 121K
R1 = 178e3;         % R59 178K
%% Figure voltages from resistors
Iuv = 10e-9;        %abs minimum 10nA
Vos_UV = Iuv*(R1+R2);    % max tolerable offset of UV% 3mV
Ruv = Vos_UV/Iuv;
UVth = R3*.5/Ruv+.5;      % 20.232 v
OVth = 0.5*(Ruv+R3)/R1;     %33.986 v
​
%% Procedure for figuring resistors
Iuv = 10e-7;        % [A] abs minimum 10nA  % 1 pick bias current
Vos_UV = 10e-3;     % [V] choose UV offset tolerance
UVth = 20.;         % [V] Under voltage threshold
OVth = 34;          % [V] Over voltage threshold
Ruv = Vos_UV/Iuv;   
R3 = Ruv*(UVth - 0.5)/0.5;   % [Ohm]
R1 = ((Ruv+R3)/OVth)*0.5;    % [Ohm]
R2 = Ruv-R1;                 % [Ohm]
Itotal = OVth/(R3+R2+R1);    % [A]
P_R3 = Itotal^2*R3;          % [W]
P_R2 = Itotal^2*R2;          % [W]
P_R1 = Itotal^2*R1;          % [W]
```

closed 2021-02-26 by npetersen2

GnomeTek, 2021-03-21:
Approved!