Severson-Group/AMDC-Hardware #165: Add extra capacitor and pull-up resistor near the reset switch to reduce resetting due to ESD.

ASHAD1, 2021-03-02:
I would also prefer to have a TVS diode near the switch.
[image]

GnomeTek, 2021-03-04:
Thinking about this, I think @ASHAD1 suggestion of having a TVS is a great idea, my comments were to be added along with that. 
In summary: 
TVS near the switch to inline resistor (10k? pick a nice common value you have) and the 100nF that is in place is adequate.

npetersen2, 2021-03-17:
@ASHAD1 @GnomeTek I think I've added what you both have recommended:

Please review.

[image]

GnomeTek, 2021-03-17:
This will work, although best practice is to place TVS as close as possible to external source. So close to SW1, on the other side of R21.

npetersen2, 2021-03-25:
Rearranged ordering of components per above comments.

See 332976bfcc88670f4fcbdf6df80678a233aa8aa7

[image]

closed 2021-03-25 by npetersen2