Severson-Group/AMDC-Hardware #168: Documenting AMDC restart issue

elsevers, 2021-03-11:
This issue is documenting observation of an EMI issue in the AMDC REV D. In this instance, the USB UART link to the PC would itself, presumably due to EMI.  

- The problem occurs when a command is issued to enable the PWM interface while two 3phase inverters are connected to the PWM ports with the DC link energized. If the DC link is at a "low" voltage, this issue never appeared (in this case, low voltage was 30V; a 50V bus caused the issue quite repeatably)

- When the PWM interface enable command would go out, the AMDC would stop responding to UART commands from the PC. However, **it did not restart itself**. The AMDC was still alive (@ASHAD1, can you confirm that the AMDC was still executing code properly? was there a heartbeat LED blinking?), but we could not communicate with it via UART. 

- To get the AMDC to communicate via UART again we restarted the AMDC and disconnected/reconnected the USB cable to the PC. We would restart the AMDC by pressing the restart button on the board. We did not have to cycle power on the AMDC.

- We were using a laptop and disconnected the laptop's power supply so that the laptop had no earth ground in it. This did not in any way change the problem. This fact, combined with the restart of the AMDC fixing the communications, leads me to conclude that it is AMDC hardware that is ceasing to function properly.

- When the PWM interface enable command went out, *I think* (@ASHAD1 please edit this text to make it correct) that all PWM duty ratios were set to a value that should cause no current to flow in the three phase inverter but that the inverter would start switching at this point. For example, maybe they were all set to 50%. 

*Brain storming on causes/solutions:* Perhaps this indicates that the current USB UART IC is overly sensitive to noise, and this problem will be resolved in REV E by usage of the new part. Alternatively, I am wondering if this issue makes a case for adding some sort of EMI protection on the PWM ports. OR, perhaps this makes the case for moving the AMDC further away from power stacks / current carrying conductors and into an EMI shielded box...

ASHAD1, 2021-03-11:
@elsevers the issue that you have described is correct, this happened before when I was testing the BP1 motor. When PWM is turned on (all duty ratio set to 50%) the UART link is disconnected, but the AMDC doesn't restart -- I can see that LED indicating PWM status was still on after the EMI event.
The event occurs at higher bus-voltage. I have noticed for bp1 this occurred at 100V bus voltage, for bless IM (Jiahao prototype) this occurred at 50V bus voltage.

npetersen2, 2022-04-19:
I am closing this issue as it has not been a problem with anyone recently, please reopen if any concerns come up.

closed 2022-04-19 by npetersen2