Severson-Group/AMDC-Hardware #169: Connect mounting holes to chassis GND, not the isolated board common

npetersen2, 2021-03-15:
All mounting holes are current connected to `GND` -- the AMDC common ground (i.e. ground plane).

[image]

If the AMDC is screwed to a metallic box with metal stand-offs (as in the BOM), this will connect the board common to the outer world... This should be avoided.

Fix this by connecting the mounting holes to `GND_IN` -- i.e. the input supply voltage source which exists BEFORE the power input stage and isolation etc.

npetersen2, 2021-03-17:
Updated all mounting holes to `GND_IN`

[image]

closed 2021-03-17 by npetersen2

GnomeTek, 2021-03-21:
Approved!