Severson-Group/AMDC-Hardware #171: Add local decoupling to VREF at diff amp ICs

npetersen2, 2021-03-15:
Looking at the layout... the VREF going to the Analog Signal Conditioning circuitry is only decoupled at the voltage reference IC itself. The signals then go a ways across the board before being used by the front-end circuitry.

Add some local decoupling to the VREF signals near the diff amps to help noise immunity since VREF is a constant value.

[image]

npetersen2, 2021-03-15:
Added one 0603 0.1uF cap for each diff amp:

[image]

closed 2021-03-15 by npetersen2

npetersen2, 2021-03-15:
Very hard to see, but I placed these caps physically close to the diff amp ICs:

[image]

GnomeTek, 2021-03-21:
Approved!