Severson-Group/AMDC-Hardware #174: Power Stack Supply Voltage Designations

GnomeTek, 2021-03-24:
Power Stack Power Supply is noted differently in different places. The biggest offender is page 1 of schematic it's referenced as VDDPS. This is what prompted the deep dive for me. 

Power Stack Supply Voltage (Not PWM drive voltage)
- VINPS - on silkscreen
- VDDPS - on schematic block diagram (1st page)
- VIN_PS - Noted in PowerDistribution.md under Power Stack heading
- VPS in powerstack.md 
- Not present in Power.schDoc
- **VIN_PS net label** 

Suggest:

- Update schematic 1st page block diagram 
      VDDPS -> VINPS  (To match silkscreen)
- Update all text references, including block diagram in PowerStack.md
      VPS -> VINPS
- Include VINPS and any other pass through/externally supplied power in PowerDistribution.md Block Diagram
- Move "Power Connections" from BoardIO to Power schematic (or link them via schematic ports. Absolutely valid to have them listed on BoardIO. but also needs to be noted on Power, even as a pass through)

Reference from VINPS to VIN_PS is trivial for a person to make. No need to update schematic net names or silkscreens. IMO.

npetersen2, 2021-03-25:
Implemented the following:
1. Changed block diagram label from `VDDPS` to `VINPS` -- this was a mistake
2. Updated `BoardIO` sheet to expose power signals as ports, and implemented connectivity in the `Power` sheet to make schematics easier to read.

Did not update the GitHub markdown docs -- we'll need to create a separate issue for this.

npetersen2, 2021-03-25:
Closing. See new issue #176 for continuation.

closed 2021-03-25 by npetersen2