Severson-Group/AMDC-Hardware #186: System will intermittently fail to properly boot from EEprom

GnomeTek, 2021-08-17:
System will intermittently fail to properly boot from EEprom, causing several power cycle to get to "catch" 

Hunch says setup and hold times are being violated after power cycle between the FPGA and the EEprom chip. Likely the EEprom takes longer to initialize to a ready state then the FPGA. The FPGA then initiates communication when the EEprom is not ready, and FPGA fails to load bitsream.

npetersen2, 2021-08-18:
I agree with this, however, the interface between the fpga and the memory chip is entirely managed on the PicoZed device. Therefore, I think it will be hard for us to adjust the interface to meet the required timing. 

As a first step, I propose we should simply lower the clock frequency used to retrieve the bitstream from the memory device. Also, we can insert timing delays into the bootloader image on the processing system to try to wait for the memory to warm up.

GnomeTek, 2021-08-18:
Oh good, we have some options for control from the FPGA. I was worried there would be none since the system is in a cold boot state. 
Let me get a scope on it and verify that is the issue at hand. I need to deploy my controller to EEprom first. Other tasks are higher priority at the moment.

npetersen2, 2022-04-19:
@GnomeTek I believe this issue to be resolved on the REV E hardware design, so I am closing. Please reopen if you find it still problematic.

closed 2022-04-19 by npetersen2