Severson-Group/AMDC-Hardware #219: Kaman I/O Board: failed Schmitt trigger IC used for glitch filter

npetersen2, 2022-09-19:
## Description

On one of the Kaman adapter boards in the lab, I found that the second stage Schmitt trigger IC had failed.

Adding this issue so others have an idea of what to look for, should they run into issues.

## Hardware Waveforms

The glitch filter on the Kaman adapter board is composed of two Schmitt triggers and two RC LPFs. The filters help remove high frequency glitches and the Schmitt triggers convert the "soft" filtered waveform back to being a sharp square wave.

The waveforms below are captured on a board where the RC LPFs were altered to be `R = 10 kOhm` and `C = 1 pF` for both filters. This results in a higher bandwidth for the second filter.

I scoped the signal chain for the `SCK` line with no external noise (i.e. no DC link PWM switching)

### 1. Input from diff I/O receiver

[image]

### 2. First RC filter output

[image]

### 3. Output of 1st Schmitt trigger

[image]

### 4. Second RC filter output

[image]

### 5. Output of 2nd Schmitt trigger

[image]

**Notice the bad waveform from the second Schmitt trigger!!**

It appears the output stage low-side driver is damaged--there are sharp rising edges, but "RC" falling edges.

After replacing the 2nd Schmitt trigger with a different IC, the output became square wave again.

codecubepi, 2023-05-19:
I am experiencing a similar issue, my second Schmitt trigger IC seems to be faulty and lose the SCLK line entirely

#### Input to first Schmitt Trigger

[image]

#### Output of second Schmitt Trigger 

[image]

npetersen2, 2023-09-28:
@codecubepi I am trying to remember what this issue is...

How does the firmware updates fix the hardware issue? I think this is related to a termination resistor (?)

codecubepi, 2023-09-28:
I think there were two related hardware issues here:

1. The trigger ICs used on that board occasionally go faulty, which we just just fixed by replacing them (I know I had to replace the IC at least once). I think that was the issue you encountered, which I guess would not be fixed by firmware, only be replacing the part.
2. The issue I mentioned was that the filtering was too aggressive for the higher SCLK frequencies we were running at in May. We fixed that by changing the RC values on the REV C Kaman Adapter board, and then making the filter delay configurable in the firmware.

npetersen2, 2023-10-08:
@codecubepi thanks.

Let's close this issue for now and re-open or create a new one if/when someone runs into a similar issue down the road.

closed 2023-10-08 by npetersen2

elsevers, 2024-04-30:
@anirudhupadhyaya has created issue #231 which seems to be documenting the same problem. He addressed it by soldering in new components. You'll notice in the pictures of his adapters board that he has a large number of fly wires soldered in. The current theory is that these wires picked up ESD which damaged the Schmitt triggers. At this point, we aren't planning any design changes in response.