Severson-Group/AMDC-Hardware #226: AMDC: No termination on incoming GPIO signals

npetersen2, 2023-09-05:
After reviewing the GPIO interface per #215, @DivyaMendpara and I found that the AMDC has no termination on the input signals for GPIO... the datasheets for the Rx and Tx drivers (https://www.ti.com/lit/ds/symlink/am26c31.pdf, https://www.ti.com/lit/ds/symlink/am26c32.pdf) say there should be termination on the receiver side, not the transmit side.

Therefore, we need to add termination to the AMDC for the incoming GPIO signals.

[image]

There are 12 total input diff pairs, meaning 12 termination resistors should be added.

To save space, Nathan recommends using a SMD resistor array, for example: https://www.digikey.com/short/mtw2d204

The size of the package should be determined based on power rating... This is probably small, so something like 0603 is fine... You'll need to translate this into the array size.. i.e., each of the 4 in the array would be 0603, so Digi-Key will tell you the total size, not the per-resistor size... 0603 resistor has 63 mW rating, and it looks like a 1206 4-array resistor has 63 mW per element. So, let's go with 1206 size 4-array.

npetersen2, 2023-09-29:
I validated the schematic changes and PCB change. The routing to get the termination resistors into the design was a bit hairy...

I reworked it a bit, making use of an internal signal layer to try and get it cleaner and more logical.

codecubepi, 2023-10-04:
@DivyaMendpara @npetersen2 

Questions on the routing (copied from my reply on #228 ):

Is there a reason this trace for GPIO4_IN3_P wraps around the via? Is this a length-match? If it's not, I think we could take the trace directly to the via (like the dashed red line). If it is a length-match, disregard.
[image]

Why is there a trace loop on GPIO1_IN2_N?
[image]

npetersen2, 2024-03-27:
Closing since this was resolved in the REV F hardware

closed 2024-03-27 by npetersen2