Severson-Group/AMDC-Hardware #227: AMDC: PWM gate signals invalid during boot-up

npetersen2, 2023-09-05:
Many (all) users of the AMDC have reported that, during boot-up, their inverter power stacks have a shoot-through event. This typically causes the user's DC link power supply to go into current limiting mode.

~~I believe the root cause of this is the FPGA output pins go into high-Z state when no bitstream is loaded. Since PWM gate signals coming out of the AMDC are driven by the FPGA pins. The 1.8V logic level output from the FPGA is level translated twice: 1.8V to 5V, then 5V to `VDRIVE`. I believe the first voltage translation (1.8V to 5V) ends up latching in the "high-Z" state, which means the final gate signal is indeterminate and often leads to shorting the inverter half bridge.~~

Before we try to fix this, we need to understand exactly what is causing this issue...

It looks like this circuitry below should be protecting us from any issues during boot-up:

[image]

It looks like the pull-down resistor `R97` should be ensuring `DRIVE_EN` is always `LOW` during boot, i.e., when `LV_PS_DRIVE_EN` is not defined. Since all the PWM gate signal outputs are enabled/disabled by `DRIVE_EN`, this signal pull-down resistor `R97` should disable everything... But, it seems this is not working....

Let's get a REV E AMDC board and probe this signal chain during boot to see what actually happens! Why doesn't this work?

Potential idea why: `LV_PS_DRIVE_EN` is actually being driven `HIGH` during boot,.... then, our pull-down `R97` wouldn't be effective... Let's probe the top side of `R97` (i.e., pin 3 on `U42`) and see if it is high or low during boot.

npetersen2, 2023-09-14:
Good debugging session with @DivyaMendpara. We determined the root of the issue.

## Testing

During boot-up, we probed the signal path seen below:

[image]

We found that node `B` was 5V, and since `ESTOP_DRIVE_EN` was high, this means that node `C` was also high. This results in the PWM gate signal outputs on the AMDC to be active...and thus can result in undefined behavior during boot-up since the FPGA pins are high-Z.

The root cause is that node `A` was measured to be 1V during boot-up. After the PS starts running, node `A` drops to 0V. Then, when the PWM output is enabled, node `A` goes to 1.8V (as expected). However, when the AMDC is reprogrammed, node `A` is a steady 1V.

## Problem

The issue is that the pull-down `R97` is not strong enough... Based on simple math, it looks like a high-Z PS pin on the board we tested seems to source about 0.1 mA, which, driving through 10k resistor, is 1V. This tricks `U42` into thinking its input is high, which causes the whole issue.

## Solution

We need node `A` to remain low (close to 0V) when 0.1 mA is pushed into it... thus, we can change R97 to 10x smaller resistance to have a stronger pull-down.

We tested this on hardware by replacing R97 with a 1kOhm resistor. As expected, node `A` voltage during boot-up was reduced to about 120 mV. This is below the "high" threshold of U42, so nodes `B` and `C` remained low! We confirmed that the PWM outputs remained low during boot-up. Problem solved!

### TLDR Solution

Change `R97` from 10kOhm to 1kOhm.

npetersen2, 2023-09-18:
The REV E BOM does not have a 1k 0603 resistor already... but it has 470 ohm 0603.

Let's use 2x 470 ohms in series for this pull-down.

npetersen2, 2023-09-29:
I checked the schematic and PCB changes around this issue---looks good!

The schematic value text was not quite in the "right place," so I had Altium replace it to be the default spot. This was a super minor change which I committed to the branch.

codecubepi, 2023-10-04:
Approved!

npetersen2, 2024-03-27:
Closing since this was resolved in the REV F hardware

closed 2024-03-27 by npetersen2