Severson-Group/AMDC-Hardware #231: U3B and U7 failure in Kaman adapter board during testing

anirudhupadhyaya, 2023-12-05:
# Context

In the ARL project for BP4 drive, we are using two Kaman eddy current sensors. During Anirudh's testing one of the Kaman demodulator boxes got  damaged, more details on this in Severson-Group/ARL-eturbo#37. When trying to re-configure the eddy current sensors it was found that the two Kaman adapter boards had also suffered some damage. It is unclear the sequence of events or cause/ effect relationship right now.

# Details

## Board 1
Schmitt trigger IC U3B was found to be damaged. Hence the SCLK and CNV signals to the demodulator is not correct, so no data is received.

[image]

### Scope captures

I and @npetersen2 probed the CNV and SCLK signals at three locations marked 1,2 and 3. The signals at 1 and 2 look as expected while the signals at 3 at the output of U3B does not look good.

_**SCLK at 1**_

This is the clock signal from AMDC after level shifting. CLK frequency - 2000 kHz. Looks good.

[image]

_**SCLK at 2**_

This should be inverted of 1 and thats what we see.

[image]. 

_**SCLK at 3**_

This signal is not as expected. Does not look like a good CLK signal.
[image]

_**CNV at 1**_
Looks good.
[image]

_**CNV at _2_**_
Should be inverted from 1. Looks good.
[image]

**_CNV at 3_**

Not looking good here. RC behavior seen, instead of a square pulse.

[image]

## Board 2

In this board we found that IC U7 was damaged. This is responsible for level shifting the SCLK and CNV signals from the AMDC. 
[image] 

### Scope capture
SCLK signal at output of U7. Not looking good. This is not as expected, hence led us to the conclusion that U7 is damaged.

[image]

### Additional point
However, Anirudh found that after fixing this part the board was not working as expected. Further debugging revealed that U3B also to be damaged like board 1.

elsevers, 2024-03-26:
**Outcome:** @anirudhupadhyaya was able to repair both boards and them working for his purposes by replacing the components described above.

@anirudhupadhyaya or @npetersen2, can one of you leave a comment on if this issue should motivate any changes to the adapter board design?

anirudhupadhyaya, 2024-04-30:
Per my understanding here, U3A and U3B (second schmitt trigger IC) have been damaged on several boards during testing apart from this instance.. There could be some benefit to investigating this failure further and identifying a better IC

closed 2024-04-30 by elsevers