Severson-Group/AMDC-Hardware #236: TestBoard: find alternative part for BNC connector

npetersen2, 2024-02-02:
The BNC connector (https://www.digikey.com/en/products/detail/te-connectivity-amp-connectors/5-1634503-1/1755969) in the BOM for the AMDC TestBoard is obsolete. We need to find a replacement.

It should be footprint compatible with the existing part.

@codecubepi when you get a chance, can you please look for one and post a reply to this issue with suitable options?

codecubepi, 2024-02-06:
@npetersen2 

## *Original Part:* TE Connectivity 5-1634503-1 [DigiKey](https://www.digikey.com/en/products/detail/te-connectivity-amp-connectors/5-1634503-1/1755969) [Mfg](https://www.te.com/usa-en/product-5-1634503-1.html)

[image]

- [x] Height from PCB 13.7mm

## *DigiKey Recommended Alternate:* Amphenol 112404 [DigiKey](https://www.digikey.com/en/products/detail/amphenol-rf/112404/1011728) 

[image]

- [x] Should be *close enough* to be footprint-compatible
- [x] Height from PCB also 13.7mm 
- [x] DigiKey lists same key specs: 50 ohm impedance, 4 GHz max freq, 500 V
- [x] DigiKey In-Stock (and Normally Stocking), Unit Price 3.68 USD

## *TE Conn. Recommended Replacement*: TE Connectivity 5413969-2 [DigiKey](https://www.digikey.com/en/products/detail/te-connectivity-amp-connectors/5413969-2/2188140) [Mfg](https://www.te.com/usa-en/product-5413969-2.html)

[image]

[image]

- [ ] This is not even as close as the Amphenol above to being footprint-compatible
- [ ] The connector height is also much taller, 20.57mm from PCB
- [x] Same key specs: 50 ohm impedance, 4 GHz max freq, 500 V
- [x] DigiKey In-Stock (and Normally Stocking), Unit Price 5.02 USD

## *My Alternate Pick*: Cinch Connectivity VB335 [DigiKey](https://www.digikey.com/en/products/detail/cinch-connectivity-solutions/VB335/4430332)

[image]

- [x] Close enough to be footprint-compatible
- [ ] Height from PCB 18.7mm
- [x] Same key specs
- [x] DigiKey In-Stock (and Normally Stocking), Unit Price 4.14 USD

codecubepi, 2024-02-06:
If I were to choose from those three alternates, I'd probably go with the Amphenol alternate recommended by DigiKey.

PR #237 (2024-02-13): TestBoard: Replace obsolete BNC part with alternative
Resolves #236
closed 2024-02-13 by npetersen2