Severson-Group/AMDC-Hardware #247: Update AMDC uInverter REV-C BOM

mohamed-dek1, 2025-01-29:
@elsevers 

##
Should I create a PR that adds the necessary 5-40 screws to the uInverter parts log? These are used to attach the Heatsinks and the MOSFETs and we currently need them as a part of [this issue.](https://github.com/Severson-Group/Lab/issues/234)

elsevers, 2025-01-31:
The micro inverter uses [MOSFET gate drivers](https://www.digikey.com/en/products/detail/littelfuse-gsd/IXDN614CI/2424705?curr=usd&utm_campaign=buynow&utm_medium=aggregator&utm_source=octopart) as switches. These gate drivers are in a TO-220 package where the body is electrically connected to pin 3. 

This means:
1. we should place an insulating thermal pad between the TO-220 package and the heat sink. [Here](https://www.digikey.com/en/products/detail/wakefield-vette/CD-02-05-220/7070008) is an example product that is appropriately sized for a TO-220 package
2. We should use a nylon bolt and nut to secure the TO-220 gate driver to the heat sink. We should use the recommended bolt/nut size for a TO-220. I think this is the 4-40.

So, please add to the BOM:
1. an insulating thermal pad
2. a 4-40 nylon screw and nut pair

elsevers, 2025-02-03:
@niloofarramroodi and @noguchi-takahiro, note that I have just updated the above comment to include a discussion thread and the possibility of using a shouldered washer.

elsevers, 2025-02-04:
See page 22 of our power stack's [manual](https://drive.google.com/open?id=1tFRP3jgKel58_gy9TGWC_Hh4EHQLk1PX&usp=drive_fs) for more guidance on this
[image]

Key items:
- use 4-40 screw and locking washer
- avoid nylon as it will melt under high temperatures

I see that you can search for insulating washers for heat sinks on [Mouser](https://hr.mouser.com/c/thermal-management/?type=Insulating%20Washer)

mohamed-dek1, 2025-02-05:
## Update

I've found the following which could be used.

- [MACHINE SCREW PAN PHILLIPS 4-40](https://www.digikey.com/en/products/detail/b-f-fastener-supply/PMS-440-0025-PH/49819)
- [HEX NUT 1/4" 4-40](https://www.digikey.com/en/products/detail/b-f-fastener-supply/HNZ-440/5738)
- [THERM PAD 17.78MMX12.7MM ORANGE](https://www.digikey.com/en/products/detail/wakefield-vette/CD-02-05-220/7070008)

I've tried looking for a set of insulating screws and a washer/nut combo but most of them were nylon. The set that I found is made of zinc. The thermal pad is the one that @elsevers sent as it is meant for the TO-220 package

mohamed-dek1, 2025-02-05:
@elsevers 
The half bridges currently use a screw, hex nut, and thermal pad to secure the heatsink.

## Front
[image]

## Top
[image]

## Back
[image]

elsevers, 2025-02-06:
Alright, it sounding like we aren't really following the best practices all around. 

Check out this kit on Mouser ([532-4880SG](https://www.mouser.com/ProductDetail/Aavid/4880SG?qs=MnLw5eI5rfiaN93%2F8XQqcQ%3D%3D) $3.75) for mounting heat sinks. This snippet from their datasheet really clearly illustrates what is going on:

[image]

Notice their use of `item 2`, which is an insulating shoulder washer / bushing which is made from Polyphenylene instead of nylon. Here is a screenshot from Digikey:

[image]

They also use a lock washer and hex nut.  Everything is for a `# 4-40` screw size.

SO, we either order 1 of these kits per heat sink or we start purchasing all of the components on that list individually.

Let's have you go ahead and order 10 of these kits [532-4880SG](https://www.mouser.com/ProductDetail/Aavid/4880SG?qs=MnLw5eI5rfiaN93%2F8XQqcQ%3D%3D) and we'll try them out on our uInverters. When you get them in and try them out, show me and we'll come up with a plan for what to put on the BOM.

elsevers, 2025-02-06:
I created an issue for purchasing these here: https://github.com/Severson-Group/Lab/issues/245

mohamed-dek1, 2025-02-14:
@elsevers, here's an update and a proposed plan for the mounting kit.

## Update
The mounting kits have arrived.
See this issue:
- https://github.com/Severson-Group/Lab/issues/245)

I've installed them onto my uInverter as shown below.

## Front
[image]

## Top
[image]

## Back
[image]

## Plan
Setup a uInverter **with** the mounting kit installed (mine) and another one **without** it (@mhmokhtarabadi). Get both hooked up to an AMDC with the same test and observe how each one does.

symen013, 2025-02-19:
@mohamed-dek1, what is the status of this issue? Is there anything I can help with?

mohamed-dek1, 2025-02-19:
@symen013 thanks for the reminder, I was waiting on a response from @elsevers on moving forward with testing the new thermal kits. If the kits work out, I will make a PR configured to close this issue that adds the components to the uInverter REVC BOM.

PR #248 (2025-03-01): AMDC uInverter REVC BOM update
This PR updates the uInverter BOM for REVC to include the insulating components found in the kit used in the following issue:

- https://github.com/Severson-Group/Lab/issues/270

I'm unsure whether or not to add designators to these components as they are not in the schematic for the uInverter. The current csv does not contain these designators.

##
This PR is configured to close #247
closed 2025-03-01 by elsevers