Severson-Group/AMDC-Hardware #250: Add a list of acceptable replacement parts for AMDC hardware to REV20231005F folder

elsevers, 2025-04-07:
# Abstract

Add a list of acceptable replacement parts for AMDC hardware to REV20231005F folder

# Context

We have recently been ordering AMDC boards and have determined acceptable replacement components for several items. We want to save this knowledge somewhere for future orders.

Here are locations where the information is spread across right now:
- https://github.com/Severson-Group/AMDC-Hardware/discussions/229
- https://github.com/severson-Group/lab/issues/247

# Approach
Close this issue with a PR that adds a markdown file `alternate-components.md` 

Please render this with out standard template: a header at the top, followed by a sentence or two explaining what the file is, followed by a big table

mohamed-dek1, 2025-04-09:
The following PR has been made to close this issue:

- #253

elsevers, 2025-04-21:
Moving back to in progress, due to changes just requested.

PR #253 (2025-04-26): Add Alternate Parts List for AMDC REV F
This PR adds the alternate components file for the AMDC REVF.

##
This PR closes #250
closed 2025-04-26 by elsevers