Severson-Group/AMDC-Hardware #254: Update AMDC REV20231005F/README.md to match template from AMDS REV20200920D/README.md

elsevers, 2025-04-15:
# Abstract

Update AMDC [REV20231005F/README.md](https://github.com/Severson-Group/AMDC-Hardware/blob/develop/REV20231005F/README.md) to match template from AMDS [REV20200920D/README.md](https://github.com/Severson-Group/AMDS/blob/v2.0.x/Mainboard/REV20200920D/README.md)

# Context

As part of ordering new hardware, we have been reviewing the information provided in our release README.md files. The outcomes of this resulted in key updates to the AMDS release README as part of this PR:
- https://github.com/Severson-Group/AMDS/pull/56

# Approach

Close this issue with a PR that update the AMDC's [REV20231005F/README.md](https://github.com/Severson-Group/AMDC-Hardware/blob/develop/REV20231005F/README.md) to use the same template as the AMDS's [REV20200920D/README.md](https://github.com/Severson-Group/AMDS/blob/v2.0.x/Mainboard/REV20200920D/README.md)

elsevers, 2025-04-15:
Make sure to confirm the minimum hole size as part of this from the Altium files.

mohamed-dek1, 2025-04-17:
The following PR has been made to close this issue:

- #255

##
> Make sure to confirm the minimum hole size as part of this from the Altium files.

I've confirmed the `Min Hole Size` and `Number of Holes` using an Altium-generated drill report file

[image]

PR #255 (2025-04-18): AMDC REVF README information standardization
This PR updates the AMDC [REV20231005F/README.md](https://github.com/Severson-Group/AMDC-Hardware/tree/develop/REV20231005F) to follow a similar format to the AMDS [REV20200920D/README.md](https://github.com/Severson-Group/AMDS/blob/v2.0.x/Mainboard/REV20200920D/README.md)

- Refactors `Board Parameters` and `Assembly Details` tables.
- Adds a `Bill of Materials` section.

##
This PR resolves #249, resolves #254
closed 2025-04-18 by Known4225