Severson-Group/AMDC-Hardware #267: Post to the discussion board to document how the eLev lab tests new AMDC boards

mohamed-dek1, 2025-07-03:
# Abstract
_To do: add one sentence description of what needs to happen_

# Context
We've recently been testing new AMDC boards and need to document how we are testing them for future reference.

- https://github.com/Severson-Group/Lab/issues/311

# Approach
- Write up a post to the discussion board that illustrates how the eLev lab tests a new AMDC board.

mohamed-dek1, 2025-07-16:
# DRAFT

TITLE: **AMDC REV F eLev Lab Testing Procedure**

##

> The following procedure documents how the eLev Lab tests new AMDC REV F boards.

# eLev Lab Testing Procedure For: AMDC REV F

## Out-the-box

1. Inspect the board for any obvious soldering issues or bridges.
    - If issues arise, try to fix them using flux and desoldering tools.
2.  Using a multimeter, inspect the board for possible shorts.
    - If issues arise, look for any `DNP` resistors that may have been populated mistakenly.
    - Inspect ICs near the short for any hardware issues.
3. Ensure all ports, DC-DC converters, and the Avnet PicoZed are populated.
4. Screw in all 13 standoffs using the ones listed in the BOM.
5. Label each board with a serial number based on the board logs.

## Bring-Up

6. Follow the instructions for the [REV F Bring-Up](https://docs.amdc.dev/hardware/revisions/rev-f/rev-f-bring-up.html).
7. Follow the instructions for [Building and Running Firmware](https://docs.amdc.dev/firmware/xilinx-tools/building-and-running-firmware.html).
8. After booting up the board, ensure that the LEDs begin blinking.
9. At this point, you should have an AMDC that can be flashed properly.

## Port Testing

### GPIO

10. Connect a working [uInverter](https://docs.amdc.dev/accessories/uinverter/index.html) and use a BNC to measure the current waveforms on an oscilloscope.

### POWER STACK | ANALOG

11. Connect a working [TestBoard](https://docs.amdc.dev/accessories/test-board/index.html).
12. Probe all PWM outputs using a logic analyzer like the Saleae, and ensure the correct signals at different switching frequencies and duty ratios.
13. Follow the [automated testing](https://github.com/Severson-Group/AMDC-Firmware/tree/develop/sdk/app_cpu1/common/usr/pcbtest) procedure to test the ADC and voltage rails.

### ENCODER

14. Connect the DB-9 from the encoder near bench 3 to the top encoder port on the AMDC.
> There is currently no firmware support for the bottom encoder port.
15. You will need to add firmware to read the encoder data in either an existing app ([blink](https://docs.amdc.dev/getting-started/tutorials/blink/index.html), [controller](https://docs.amdc.dev/getting-started/tutorials/vsi/index.html), etc) or create a [custom user app](https://docs.amdc.dev/firmware/xilinx-tools/create-user-app.html).
16. Rotate the encoder one revolution and poll the sensor.
17. Rotate again and ensure a new reading is polled.

### ETHERNET

18. Follow the [Host Interface](https://docs.amdc.dev/getting-started/user-guide/host-interface/index.html) instructions to setup the network adapter for the AMDC.
19. Read the [Python Wrapper](https://docs.amdc.dev/getting-started/user-guide/host-interface/python-wrapper.html) documentation to understand how to setup up a [Programmatic Interface With Python](https://docs.amdc.dev/getting-started/user-guide/host-interface/index.html#programmatic-interface-with-python).

## Optional

20. Switch the POWERSTACK `status` lines from `output` to `input` in hardware (jumpers) and firmware `sts_mux.h` and validate status of pins.
21. Remove the jumper for the `ESTOP` and see if the PWM output is disabled

## Resources

- [AMDC REV F Release Folder](https://github.com/Severson-Group/AMDC-Hardware/tree/develop/REV20231005F)
- [AMDC REV F docs](https://docs.amdc.dev/hardware/revisions/rev-f/index.html)

mohamed-dek1, 2025-07-16:
@knvprasad-dr, can you review this draft discussion post.

knvprasad-dr, 2025-07-17:
Per 7/17: @mohamed-dek1, this looks good to me. However, I have one suggestion to add some pictures to see what is expected at each important step.